Seagate/ToolBin #3: Seagate Constellation ES SAS drive firmware update fails

Dead2, 2017-12-05:
I have bought 10x 2TB sas drives, and was hoping to use them with my Areca raid controller, unfortunately the controller keeps throwing out random disks during raid build, usually before 50% has passed, two disks have already failed. So I thought a firmware upgrade might help, and the support site does have a new firmware available labeled "Important". Current drive is 0006 and new is 0008.

I am running an RHEL 7 clone, with kernel 4.9.63 and the disks are connected to an mvsas controller (AOC-SASLP-MV8). The utils are able to talk to the drives just fine through the /dev/sg5-8 devices, but the firmware upload fails on all 4 drives I have tested on so far.

I have tested using sg_write_buffer as well, resulting in the following error:
`# sg_write_buffer -v --in=MU-SAS-0008.LOD --length=1107456 --mode=5 /dev/sg5
    Write buffer cmd: 3b 05 00 00 00 00 10 e6 00 00
write buffer: pass through os error: Invalid argument
Write buffer failed res=-1`

See gist for disk drive info and firmware error log
https://gist.github.com/Dead2/60037f2a75fa1253cbb13b8ba05e92ee

My skill level with Linux and hardware/software is high, and firmware updates like this is usually a breeze even through JTAG, but this time I am completely lost, partly because the firmware util does not really give me any clues whether this is OS, Driver, HBA or Drive related. I also found no DOS utils or bootable iso, the only alternative being windows utils, but that is a whole other mess.

If you have any ideas or tips that can help me make use of these drives, I'd be very grateful.

rudock1, 2017-12-07:
This may be difficult to solve.

Based on the error reported at the end of the log, it looks like the drive is rejecting the new firmware because it thinks it is incompatible. Drives from this time often have capacity specific versions, like a 500GB 0008 file has a different servo than a 1TB or 2TB 0008 (your ST32000444SS).

The SCSI code is 0008 in all cases but the combined servo code is changing based on the capacity (number of platters inside the drive). It may be very difficult to track down the correct firmware for this older drive. Also, if the drive was originally sold by an OEM then it might even have their specific FW and only the OEM has the updates to their own code.

from your log:
Sense Key: 5h = Illegal Request
ASC & ASCQ: 26h - 99h = Vendor specific ascq code
FRU: 6h = Vendor Specific

05 26 99 06

05 Illegal Request
26 Invalid parameter or field or other attribute
99 Invalid Field Parameter—Firmware Tag
06 FRU unknown to me, but might mean a mismatch between the servo on the drive and in the new code file

https://en.wikipedia.org/wiki/Key_Code_Qualifier
http://www.t10.org/lists/asc-num.htm
https://www.seagate.com/files/staticfiles/support/docs/manual/Interface%20manuals/100293068j.pdf
see 2.4.1.6Additional Sense and Additional Sense Qualifier codes

Dead2, 2017-12-07:
The firmware file was downloaded from the seagate support site after searching for files associated with that specific drive's serial number.

Looks like I might have bought junk though, the smart data has been wiped from the disk, thus having them show a fake 0 hours runtime, but SeaChest shows 2-4 years of runtime for them. I guess there is no way to know just how broken these disks really are at this point.

Turns out when they are plugged in with an LSI sas2 expander, the expander does not boot. When connected directly to the isci sas controller, it will not find the disks at all, it only reports that the cables are short. And the mvsas controller will not post with the drives connected, but will operate on them when hotplugged. The Areca shows none of these problems, and just throws them out at random.

So unless I find some cure within a couple of days, I guess I'll face facts and deliver them to recycling.

Thanks for your help diagnosing the problem though, the tools are awesome, and I will probably be recommending Seagate disks next time we need new drives at work 👍

xahmad, 2017-12-18:
Thanks for the feedback.

closed 2017-12-18 by xahmad