Seagate/ToolBin #5: formatUnit or fastFormat recommended?

tom1v, 2018-11-13:
I really like the extensive documentation for the SeaChest tools, but for the Format utility it doesn't mention whether formatUnit or fastFormat is recommended when changing the sector size from 512e to 4096.
I have a pair of ST12000NM0027 drives that came formatted as 512e and I'd like to reformat them to 4K. I would do a long format if it provides any benefit over fast format, even if it takes days to complete.
Please advise. Thanks!

rudock1, 2018-11-19:
sorry for the delay in replying to your question.  I created few paragraphs describing FastFormat which I will  be adding to the user guides.  I hope I covered the topic well enough to answer your questions.  If not, please to let me know what needs to be addressed.  Thanks so much for your comments!

Here is the new section:
================
About FastFormat
================
A FastFormat (quickly changing the logical sector size) may take a few minutes for the process to complete. The disk activity LED may show activity during the conversion.  The larger the drive, the longer it takes.  This feature is found on SAS and SATA interface Seagate drive families designed and manufactured later than 2016.  SAS interface drives will report support for FastFormat if it is available in the drive. Just run the -i or --deviceInfo command and check under the Features Supported section.  Unfortunately, SATA interface drives do not report this capability.  Please refer to the SATA product manual for verification of this feature.

SAS and SATA interface drives accomplish a FastFormat this with the --setSectorSize [new sector size] command found in SeaChest_Format or SeaChest_Lite.  In addition, for SAS interface drives, only, the same result is possible using the --formatUnit command with the --fastFormat [fast format mode] modifier equal to 1.  This command is found in SeaChest_Format and SeaChest_Erase.  For example:
--formatUnit 4096 --fastFormat 1 --confirm I-... etc

SAS interface drives, only, can all do a long format unit.  This procedure takes a very long time, more than a day in some cases, but it does have the advantage of full defect management with all sectors being readable at the conclusion.  The fast format does not do this full drive read and verification.  If your drive is new, there is no real need to run a long format unit to change sector between 5xxE (512 bytes per logical sector emulation) and 4kN (4096 native bytes per sector, logical and physical).

NOTE: Operating systems do a device discovery during start up and set various parameters, like total sectors and sector size, into the storage device descriptions. The logical sector size times the number or logical sectors defines the drive capacity.  You should expect to see OS I/O errors if you change the logical sector size on a drive and then perform read or write operations before the OS has updated its storage device descriptions.  Some operating systems will throw an error after accessing a drive that has just run a FastFormat but during its error recovery routines it may re-discover the device parameters and update the system logs.  A system restart, however, is the most reliable way to refresh the storage device descriptions.

closed 2018-11-19 by rudock1

tom1v, 2018-11-19:
Thank you! This is exactly the information that I needed: there's no 
point in doing a long format on new SAS drives.

Tamas