Seagate/ToolBin #7: --setAPMLevel "non-volatile"?

obetz, 2020-02-24:
SeaChest_PowerControl.1100.txt writes about the --setAPMLevel option:

"This setting is non-volatile so it holds across power cycles"

In my case (ST2000LM015, Windows 10 1903), the APM level is reset to 128 after sleep mode, so this might be a documentation issue.

maenpaa24, 2020-05-23:
I suffer from roughly the same issue with an ST2000LM010 in both Windows 10 and Linux. The difference is that the APM level is reseted after a reboot. I mean, it survives a sleep command.

I would like Seagate to make the --setAPMLevel command non-volatile.

sakshatshinde, 2020-05-28:
```
SATA Only
=========
	--showAPMLevel   (SATA Only)
		Use this option to show the current Advanced Power Management
		(APM) level of a device

	--setAPMLevel [1 - 254]   (SATA Only)
		Use this option to set the APM level of a device. Valid values
		are between 1 and 254.  This setting is non-volatile so it
		holds across power cycles.
		      1 = Minimum power consumption with standby mode
		  2-127 = Intermediate power management with standby mode
		    128 = Minimum power consumption without standby mode
		129-253 = Intermediate power management without standby mode
		    254 = Maximum Performance
```
I believe this is the part of documentation which is mentioned. 

> The term power cycle, or sometimes called power cycling, refers to the act of turning off the computer and turning it back on immediately

I suspect that it might be valid for a certain period of time, maybe? The wording here is a bit unclear. Does it only last for a certain amount of time? 
But then again the above people mentioned that for one it doesn't hold for sleep and for other it does. Which makes it more unreliable. 

Regardless an update on this would be great.
For now @obetz @maenpaa24  might try writing a startup script to execute this command.

rudock1, 2020-05-29:
Hi @obetz @maenpaa24 and @sakshatshinde ,

Sorry it has taken so long for this question to get answered.  
It is a documentation error to say that "This setting is non-volatile so it holds across power cycles".
It will be changed to "This setting is volatile so it does not hold across power cycles".

Our apologies for the confusion this has caused and for time lost to debugging.
Documentation updates will be pushed very soon.

closed 2020-05-29 by rudock1

rudock1, 2020-05-29:
Some additional information from a subject matter expert:

Hi everyone,
The APM (Advanced Power Management) feature on SATA drives is defined as a volatile feature in the T13 specifications.

It is not possible to make it hold to a set value on reboot from drive commands themselves.
Other tools, such as smartmontools see the same thing with APM.

If you want to configure power features that can hold across a power cycle, use the EPC (Extended Power Conditions) feature which is capable of controlling all the different power modes a drive can enter and is defined to allow all settings to be non-volatile. This is a newer feature, so unfortunately not all drives will support this.

The next best solution is to use a script that runs on reboot to set the preferred APM settings when your computer starts.