Seagate/ToolBin #8: openSeaChest binaries don't run

edrozenberg, 2020-03-11:
The openSeaChest binaries don't run for me (Slackware systems) and I'm not able to figure out the reason yet.

```
# ./openSeaChest_Basics_280_1192_64
-bash: ./openSeaChest_Basics_280_1192_64: No such file or directory

# strace openSeaChest_Basics_280_1192_64 
strace: Can't stat 'openSeaChest_Basics_280_1192_64': No such file or directory

# gdb openSeaChest_Basics_280_1192_64
GNU gdb (GDB) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-slackware-linux".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from openSeaChest_Basics_280_1192_64...
(No debugging symbols found in openSeaChest_Basics_280_1192_64)
(gdb) r
Starting program: /tmp/openSeaChest_Basics_280_1192_64 
/bin/bash: /tmp/openSeaChest_Basics_280_1192_64: No such file or directory
During startup program exited with code 127.
```

I've tried on a few different systems, and also tried an older set of openSeaChest binaries with the same results. Very strange.

No problems at all with the SeaChest binaries.

edrozenberg, 2020-03-11:
I tried building openSeaChest utilities from the source but there's an issue with the source currently (GitHub issue already exists for that problem) so am not able to build them.

For now I can use the regular SeaChest binaries and that works fine.

edrozenberg, 2020-03-11:
If I use the binaries from https://github.com/Seagate/openSeaChest/releases, those work fine. I was also able to compile my own from the openSeaChest source code (after including all the git submodules `git submodule update --init --recursive`).

MrM40, 2020-12-09:
I get the same problem (Debian 10) from the binaries @ https://github.com/Seagate/ToolBin/tree/master/openSeaChest/bin-build

Please fix, no reason to have people wast there time

MrM40, 2020-12-09:
And https://github.com/Seagate/openSeaChest/releases doesn't include the newest version 20.03.11 !?

vonericsen, 2021-03-12:
I'm sorry this has taken so long to review and update...it's been a hectic year for us.

Today I pushed updated binaries for Windows and Linux x86_64. Please give them a try and let me know if those work or not.

Linux tool distribution is less clear or easy than Windows and sometimes binaries don't run as expected. What I have found is generally, if we build in CentOS/redhat they will work with forward compatibility in other OS's. This is not always the case, but this is what we attempt to do.

If the new binaries still don't work, I will investigate further to make sure there is no funny business going on with git and compression or anything like that which I think is unlikely the problem, but it could be a problem.

vonericsen, 2021-04-28:
Closing this since I have not heard feedback since the last set of binaries were pushed.
If you encounter this issue again, please reopen this or open a new issue and we will look into it again. Thanks!

closed 2021-04-28 by vonericsen