Seagate/ToolBin #9: Cannot change logical sector size of SATA-connected Seagate: "Setting sector size not supported on this device"

mindfulsource, 2020-05-25:
I just bought an 8TB ST8000DM004-2CX188 5400 RPM Seagate HDD and I have an important reason to want to change the logical sector size to 4096 bytes. I'm on Debian-based Linux.

SeaChest tools show for my drive:

```
Logical Sector Size (B): 512
Physical Sector Size (B): 4096
```

The result of the following two commands,

`SeaChest_Format -d /dev/sg1 --setSectorSize 4096 --confirm this-will-erase-data`:

`SeaChest_Lite -d /dev/sg1 --setSectorSize 4096 --confirm this-will-erase-data`

is:

`Setting sector size not supported on this device`

Seagate website says no firmware update is available for the drive. 

Surely if the physical sector size is 4096, SeaChest should allow me to set logical sector size to 4096 somehow?

Thanks for any help.

lucipacurar, 2020-05-30:
I have the same problem with a 3TB ST3000DM008-2DM166 and don't know how to convert it to native 4K sectors.

mindfulsource, 2020-05-30:
I have now basically learned, from extensive research, that my particular Seagate drive is hard-set in firmware to be "at 512-byte sectors", and cannot be changed. (apart from perhaps firmware modification, but maybe also its physical design isn't rated to be native 4096.)

If it doesn't say '4kn' or you find a datasheet confirming, assume like my disk that it's a '512e' drive. Google more and you'll learn about what those mean.

In order to make a 512e drive change to 4k logical sector size, you'll have to find an obscure SATA-to-USB adapter, PCI SATA card (controller), or SATA USB HDD enclosure, which happens to have chipset firmware that forces the drive to present itself as having 4k logical sectors to any operating system, no matter the BIOS.

(See [here](https://forums.anandtech.com/threads/current-usb-3-x-bridge-chipsets-that-support-4096-sector-re-emulation.2567441/).)

I'm hunting down a model myself right now, I have so far bought and returned two models which advertise "8TB" drive + Windows XP compatibility (as a way to test it), but don't actually work.

I'm getting closer to finding an adapter and emailing manufacturers, and I'm determined to find something that works for me. But this type of SATA-controller hardware is virtually non-existant by 2020, so if you really want 4k logical sectors going forward you need to choose a '4Kn drive' and spend the extra bucks.

lucipacurar, 2020-05-30:
Well, I'm using my drive inside a Synology NAS, and Synology says:

> Always use and manage 4K native (4Kn) hard drives separately.
4Kn hard drives are supported from DSM 5.1-5021. Use 4Kn hard drives separately from hard drives in other disk sector formats when creating, repairing, expanding and migrating any volume, disk group, RAID group and LUN.

I was planning on buying another 3TB drive to pair them up inside the NAS. But now I guess I have to sell this one and buy a pair of 4Kn drives.

lucipacurar, 2020-05-30:
https://www.seagate.com/www-content/product-content/ironwolf/en-us/docs/100807039h.pdf

It looks like all new IronWolf drives are 512Ke, so no need to worry in the end.

jdrch, 2020-06-14:
> If it doesn't say '4kn' or you find a datasheet confirming, assume like my disk that it's a '512e' drive.

TIL. I was wondering why Google results for 512e to 4Kn conversions were so sparse. FWIW I'm starting to switch to 4Kn completely, starting with ordering a 14 TB 4Kn Toshiba MG07 yesterday.

vonericsen, 2020-08-25:
Hello everyone,
Let me start by saying that this can be confusing topic, so bear with me as I try to explain it.
This Github repo is maintained by Seagate'stools team and product specific information is expertise of Field & Customer Support Engineers, so please reach out to support team for further assistance.

The ability to switch between a 4k and 512 logical sector size requires the firmware to allow this to happen, which is what SeaChest checks for before sending these commands.
Prior to the introduction of this capability in newer products, you had to purchase a drive formatted with what you wanted:
- 512N(ative): 512 logical & 512 physical
- 512E(mulated): 512 logical & 4096 physical
- 4KN(ative): 4k logical & 4k physical

I believe the first product line this feature was enabled on Seagate devices was 12 TB drives receiving this feature. SAS has had this a bit longer on mission critical products (10k/15k RPM products).
If you are purchasing a drive and want this capability, you can look for "Fast Format" as this is typically how it is described, at least in Seagate's marketing.

If you already have a drive, running `SeaChest_Format -d myDriveHandle -i` and looking in the `Features Supported` Section can tell if this is supported or not.
 - SATA will list this as `Set Sector Configuration`
 - SAS will list this as `Fast Format`. Note: Check the product manual on SAS products as this is not as easy to detect support for.

As for those USB adapters that show up as 4KN, this is a reverse-emulation that doesn't actually do anything to the drive. If you read 1 of the USB sectors, this is translated into a read of 8 SATA drive sectors. The SAT spec mentions this is allowed, but does not set any requirements on LBA alignment, meaning the USB adapter is free to set its LBA 0 at a different place from the drive's LBA0-7, but in practice I have not seen this happen. These adapters may also be more rare as they were designed to get around a 32bit LBA limit in Windows XP, which is no longer as popular of a problem as it used to be.

closed 2021-03-12 by vonericsen