Seagate/ToolBin #10: USB Seagate External Hard Drive 16TB (STEB16000400) bricked after setSectorSize

mlogic, 2020-10-02:
On a Ubuntu 20.10 (running Linux 5.8) system, I used SeaChest Lite (downloaded from official website on 9/30/2020) and set a USB Seagate External Hard Drive 16TB (STEB16000400) to sector size 4096. The operation succeeded with no error, but the drive became sorta bricked. Now the system can't boot when the USB HDD is attached, because it kinda froze on detecting that drive. The drive's blue light would always blink, with no apparent head seek could be heard.

I could remove the drive and boot up the system normally. Then I plugged in the HDD. My Linux system recognized it as a 2 TB disk drive but any operation will block forever. I can't run smartctl or seachest on it any more.

I also tried SeaTool bootable USB. The drive would become offline (as noted in dmesg) if I issue any command. I tried Set Max LBA but the command failed.

Also tested it on another machine running Windows 10. After a very long wait, Windows recognized it as a 1.8 TB drive but couldn't do anything about it. SeaTools for Windows took a very long time to scan the disks (half an hour) but couldn't recognize it either.

Is this drive dead? Anyway to bring it back? If setSectorSize bad for USB attached drives? If so why am I allowed to issue it?

PS. FWIW, uas was disabled for that HDD on Linux so that I could use smartctl to test it before I ran seachest lite The long test finished without any error.

vonericsen, 2020-10-02:
Hi @mlogic,
Thanks for the report. I'm one of the software engineers that works on SeaChest and do not have product specific information. I recommend contacting Seagate's customer support to see if there is anything that can be done to assist you. 📞 

I use Ubuntu 20.04 as well for most of the development on our Linux tools. 👨‍💻 
The best bet to fix it is to change it back to 512B sector size using the same command setSectorSize option, but specifying 512 instead of 4096.

If you were recommended to change to 4k somewhere, can you please share where you found that information. I want to make sure this doesn't happen to others and we can put together some better information about this feature set of the drives. 📰 

The commands to change the sector size reformat the drive quickly, but if interrupted for any reason can become unresponsive or have other issues. This command set is made to allow customers to setup drives before integrating them into their environment, before any data is written to them, but it's purpose is really meant for advanced configurations in large scale storage. There is no real benefit to switching to 4k at home, especially on USB drives. I will add an additional warning to SeaChest_Lite ahead of this operation to help warn about this kind of issue.

From the sounds of it, the tool returned with success, which is a good sign that the drive itself probably isn't dead, but the USB adapter may not know how to handle the 4k block size of the internal drive now that this has been done.
Some USB adapters do not properly handle changes like this, even after power cycling them, but some of them do. It's not possible to tell which one responds well and which one doesn't.
Part of the reason for this is that back in the Windows XP days, many implemented a emulated 4k sector size to the host, but that code may still live inside current adapters. That adapter code expects the drive to have a 512B sector size, anything else will cause it to become unresponsive to the host system.

**So how do you determine whether it's the drive or the adapter having an issue?** Well you would have to break them apart and plug the drive into a SATA interface to test it and find out, but I **do NOT recommend** you do that right now. I would contact Seagate support to see if anything can be done to assist you before you attempt this.

Sorry that you are dealing with this, but thank you for reporting it to us.

mlogic, 2020-10-02:
Thanks for your prompt response, @vonericsen !

> The best bet to fix it is to change it back to 512B sector size using the same command setSectorSize option, but specifying 512 instead of 4096.

I have tried various combination of seachest and OSes but none could set the sector size back to 512B. SeaChest --info could show certain fields, such as serial number, but most feature related fields are "not supported". It seems to me that the drive only responses to a very limited set of commands in this state.

> If you were recommended to change to 4k somewhere, can you please share where you found that information. I want to make sure this doesn't happen to others and we can put together some better information about this feature set of the drives. 

There's not a single source that told me to do this. It's more like a standard procedure that we do for all newly acquired HDDs. This is the first time I do this for a USB external HDD though. I did some research before running setSectorSize and found that someone mentioned that they've successfully tried this on certain WD external HDDs. Because I didn't find anything that said this is forbidden, so I went ahead and tried that.

>  I will add an additional warning to SeaChest_Lite ahead of this operation to help warn about this kind of issue.

I agree that this would be very helpful. May just as well forbid it if the USB enclosure doesn't support 4K sector size.

> From the sounds of it, the tool returned with success, which is a good sign that the drive itself probably isn't dead, but the USB adapter may not know how to handle the 4k block size of the internal drive now that this has been done.

That's what I guessed too.

> Some USB adapters do not properly handle changes like this, even after power cycling them, but some of them do. It's not possible to tell which one responds well and which one doesn't.
> Part of the [...]

Thanks for the explanation. Sounds exactly like what's happening here.

> Sorry that you are dealing with this, but thank you for reporting it to us.

Glad this report is useful. It would be great if some warnings can be added to SeaChest so others know this is not supported.

vonericsen, 2021-04-28:
Hi @mlogic,
We added lots of warnings in the last two versions we've published on Toolbin and to our website.
Looking back at this, I think it may have been related to #16, but did not realize it at the time.
We've since added more improvements to the tools to help mitigate this kind of problem and to make sure the drive will be left in a good state even if it is unsuccessful. Please review that issue, as well as Seagate/openseaChest#54 for more information.

If you have more issues related to this, please post them and we'll do our best to take care of them, otherwise close this if you think this helps resolve this issue.

closed 2023-03-06 by vonericsen