Seagate/ToolBin #11: Not an NVMe device, Skipping

NavCC, 2020-11-05:
Hey Guys,

Executable: SeaChest_NVMe_111_12130_64.exe
Version: v1.1.1

Having trouble with this one SSD brand, other PC's with different SSD's are fine.

[nvmehealth.txt](https://github.com/Seagate/ToolBin/files/5492338/nvmehealth.txt)

I've attached the log, the scan shows the device as NVMe, which it is but trying to retrieve specific information returns that it is not an nvme device.

Thank you for your tool, has been very useful.

vonericsen, 2020-11-05:
Hi @NavCC,
Thanks for the report.

This looks like an error in detecting it as NVMe.
Can you confirm how this is attached to the PC? M.2/PCIe slot or USB, etc?
Also, would you mind attaching a verbose log. You can do this by adding `-v 4` to your command line. You will probably need to output it to a file as this will print all the commands and returned raw data from the device which I want to check to see if any of that is doing something odd or unexpected.

One last thing, if you run the `-i` option in another SeaChest tool (like SeaChest_Basics), what does it output?
If you include both regular and verbose output this may help me determine what the issue is.
`SeaChest_Basics -d PD? -i > basicsInfo.txt` & `SeaChest_Basics -d PD? -i -v 4 > basicsVerboseInfo.txt`
SeaChest_NVMe has a check in it to ensure it is only used on NVMe devices, so other tools may be able to proceed and show information about the drive.

NOTE: On USB attachment, all devices show up on the system as SCSI devices as that is how the interface standards are defined and we would need to be able to perform pass-through commands on that adapter to find NVMe (or SATA) drives when connected this way, and not all of these devices are supported.

NavCC, 2020-11-06:
Hello @vonericsen thanks for your prompt reply, I've collated the following information on the device in question.

Dell OptiPlex 3070.
The M.2 drive connects directly to the motherboard.
The M.2 drive is a Skhynix
Part: HFM256GDHTNI-82A0A 
Model: BC511
Interface: PCIe Gen3 M.2
Capacity: 256GB 

I've also attached the requested logs but they more or less say the same thing.
[basicscaninfo.txt](https://github.com/Seagate/ToolBin/files/5498344/basicscaninfo.txt)
[basicsVerboseInfo.txt](https://github.com/Seagate/ToolBin/files/5498331/basicsVerboseInfo.txt)
[VerboseInfoSeaChest_nvme.txt](https://github.com/Seagate/ToolBin/files/5498332/VerboseInfoSeaChest_nvme.txt)
[basicsInfo.txt](https://github.com/Seagate/ToolBin/files/5498333/basicsInfo.txt)

Please let me know if anything else is required.

vonericsen, 2020-11-06:
Hi @NavCC,
It looks like there was an error with the drive handle you provided in the logs.
Based on your scan, it looks like you need `PD0` but you tried `PD1` and `PD01` which aren't available devices.
Would you please retest again using `PD0`?
Thanks

NavCC, 2020-11-09:
Hello @vonericsen 
Woops yes I can see that now.

[basicsVerboseInfo.txt](https://github.com/Seagate/ToolBin/files/5507664/basicsVerboseInfo.txt)
[seachest_nvmverboseinfo.txt](https://github.com/Seagate/ToolBin/files/5507665/seachest_nvmverboseinfo.txt)
[Seachestnvmescan.txt](https://github.com/Seagate/ToolBin/files/5507666/Seachestnvmescan.txt)
[basicsInfo.txt](https://github.com/Seagate/ToolBin/files/5507667/basicsInfo.txt)

Have done the requested again, looks like it's sending SCSI requests.

Just spotted in device manager it's being recognized as a SCSI device, perhaps something to do with the IRST driver although it's not in a RAID.

In device manager: SCSI\DISK&VEN_NVME&PROD_BC511_NVME_SK_HY\4&4E9F7D6&0&010000

Updated the device to latest firmware, no difference.
I think this is the way Dell has configured these devices, i've not seen this behavior before.
When I saw Vendor NVMe in the SeaChest_NVMe tool I wrongly assumed it was configured as nvme storage.

vonericsen, 2020-11-09:
Hi @NavCC 
Thanks for the updated logs. It is definitely seeing it as a SCSI device, likely due to the driver exposing it that way.
I've previously added some support for some drivers to pass-through NVMe commands when this happens, but it doesn't appear to fix all situations so far. I added it for the open-fabrics NVMe driver interface (which different vendors may or may not use) as well as one method for talking through some Intel drivers (but they are not all the same).

If you use the tool Crystal Disk Info (opensource, free tool), what does it show about the drive? 
Here is an example from a system drive I have, but this is not using the same IRST driver like yours is...just looking for a comparison.
[image]

Let me see if we can order one of these systems so I can do some hands-on testing and see if there is a way to access it with NVMe commands some other way. On your system, did you do any kind of system configuration to see this drive like this, or is this more or less a "straight out of the box" setup?

NavCC, 2020-11-10:
Hello @vonericsen 

As far as I know, is an out of the box configuration, we have not done anything funky but if you look at Dells driver and support page for this product it looks like it could have come with a number of different nvme drives.
[OptiPlex 3070 Drivers](https://www.dell.com/support/home/en-au/product-support/product/optiplex-3070-desktop/drivers)
Perhaps this was done for compatibility reasons as Dell only lists firmware updates for the nvme drives and not drivers.

[image] Screenshot requested.

Thank you.

vonericsen, 2020-11-10:
Hi @NavCC,
Thanks for the additional info!
The NVMe system driver may not have an update from Dell, but it looks like some drives that come in the system may have vendor specific firmware updates. I believe they would all show up the same way regardless of which one it is.

It looks like crystal disk info knows how to pass-through commands to your NVMe drive. It was able to get the serial number and the attributes it listed which can only come through direct NVMe commands, not SCSI translation that is otherwise in use.
I believe I have this implemented, but have not released it in a new executable yet. It is part of the openSeaChest code though.

We are due for an update and I am beginning that process now, but not sure exactly when we will publish the update.
When I am closer or have completed the release, I can add another comment on this issue to see if that new version fixes the issue.

If you are able to compile openSeaChest, you would be able to test the additional code I added and believe may work. Unfortunately I haven't had a system like this to test it directly, so I haven't been able to perform complete testing on it. The closest I've been able to test so far is an optane boost system which works similarly, but I know even that is still a little bit different.
We are working on ordering one of these systems so that we can ensure it works in the future since we want to be able to support this scenario as well.

NavCC, 2020-11-11:
Hi @vonericsen 

My Pleasure :) 
Unfortunately after compiling the dev branch to make OpenSeatChest_NVMe Version 2.0.0-2_1_0 it was still unable to work.

I've attached the logs both with the -i flag and verbose 4 if you wanted to look over them. 

[OpenSeatChest_basics.txt](https://github.com/Seagate/ToolBin/files/5521989/OpenSeatChest_basics.txt)
[openseachest_NVMe.txt](https://github.com/Seagate/ToolBin/files/5521990/openseachest_NVMe.txt)
[OpenSeatChest_basicsverbose.txt](https://github.com/Seagate/ToolBin/files/5521991/OpenSeatChest_basicsverbose.txt)
[openseachest_NVMeverbose.txt](https://github.com/Seagate/ToolBin/files/5521992/openseachest_NVMeverbose.txt)

If there is anything additionally you would like to try I'm more than happy to run tests on this system.

vonericsen, 2020-11-16:
Hi @NavCC,
Thanks for all the testing!
I'm sorry that it didn't work. I don't have anything else to try right now. Once we get a machine in to sit and debug, we'll find a solution.

This issue will be left open until it is resolved and you've had a chance to test it as well.

vonericsen, 2021-01-21:
Hi @NavCC,

Just wanted to provide a quick update: We've received a Dell Optiplex 3070 and I can repeat the issue. 
I've completed an initial debug of what is happening and now know why we aren't discovering this drive as NVMe.
This drive isn't showing up as expected. The driver is reporting it as RAID mode (even if it isn't in RAID), so the current code to check for additional NVMe capabilities isn't being run.
Even after some quick hacks to attempt it, it still isn't functioning quite as expected, but I believe I can get it working. I need to do more debugging and testing to figure out the best solution to support this. Once I've figured out a solution and it passes some testing on the systems I have, I'll update this issue again.

vonericsen, 2021-03-12:
Hi @NavCC,

I have pushed updated binaries here and to openSeaChest that should correct this issue.
Please give them a try and if everything looks good, close this issue.
If you have any problems, please let us know!

NavCC, 2021-03-15:
Hello @vonericsen 

Thank you very much for your updated binaries, I had a chance to test them today and was able to retrieve the information i required!

Thanks for sorting this issue out, would imagine it would effect a fair few devices if the Intel RST was in use.

Have a wonderful day :)

closed 2021-03-15 by NavCC