Seagate/ToolBin #12: SeaChest_Lite_120_11923_64.exe: unknown option -- showSupportedSectorSizes

mvadu, 2020-11-26:
1. Downloaded latest seachest zip from http://support.seagate.com/seachest/SeaChestUtilities.zip
2. Extracted, and ran SeaChest_Lite_120_11923_64.exe --scan, found the device i am interested in
3. Trying `SeaChest_Lite_120_11923_64.exe -d PD3 --showSupportedSectorSizes` to find if the HDD supports 4096 sector size natively
```
SeaChest_Lite_120_11923_64.exe -d PD3 --showSupportedSectorSizes
SeaChest_Lite_120_11923_64.exe: unknown option -- showSupportedSectorSizes
SeaChest_Lite: Unable to parse --showSupportedSectorSizes command line option
Please use --help for more information.
```
4. Help actually shows `showSupportedSectorSizes` is indeed an supported option.
```
--showSupportedFormats
                This option will show the supported formats of a device.
                These can be used to change the sector size or
                used with a format operation. On SAS, this is the
                supported block lengths and protection types VPD page. (SBC4
                and later) On SATA, this is the sector configuration log. (ACS4
                and later) If the device does not report supported sector
                sizes, please consult your product manual.
```

vonericsen, 2020-11-30:
Hi @mvadu,

I downloaded the package and tested it and see the same thing.

Looking through some changes and verifying the help output from the tool (not the user guides), the `--showSupportedSectorSizes` was replaced with `--showSupportedFormats` to consolidate a couple different outputs into one option.

It looks like the user guides are still referencing `--showSupportedSectorSizes`, so I will work on updating those to correct that issue.

Let me know if you have any other trouble with this.

vonericsen, 2021-04-28:
Hi @mvadu,

This should have been corrected in the last update published to the website earlier this month and the last two versions pushed here.
We are still in progress posting the most recent version (v21.04.27) to the website, but it is available on Toolbin already as of today.
Please close this issue if it resolved this for you, or please let us know if you have any additional issues regarding the documentation. Thanks!

closed 2023-03-06 by vonericsen