Seagate/ToolBin #21: Drive unusable after interrupted Fastformat

JSinghDev, 2021-12-10:
I was trying to change my ST4000NM005A SAS drive from 512e to 4kn and I ran the command: 
SeaChest_Format_x64_windows_R.exe -d arc:0:0:4 --formatUnit 4096 --fastFormat 0

I realized that I should have entered 1 for fastformat paramter instead of 0 and now the drive features look like this: 

Protection Type 1
Protection Type 2
Self Test
Automatic Write Reassignment [Enabled]
Automatic Read Reassignment [Enabled]
EPC
Informational Exceptions [Mode 6]

**Earlier it used to look like this:** 

Protection Type 1
Protection Type 2
Application Client Logging
Self Test
Automatic Write Reassignment [Enabled]
Automatic Read Reassignment [Enabled]
EPC
Informational Exceptions [Mode 6]
Translate Address
Rebuild Assist
Seagate Remanufacture
Seagate In Drive Diagnostics (IDD)
Format Unit
Fast Format
Sanitize

-------------------

I cannot run the formatunit command again, Is there any way to restore the drive to factory defaults

vonericsen, 2021-12-13:
Can you share the output of the following options?
`SeaChest_Format_x64_windows_R.exe -d yourDevice -i --showSupportedFormats`

JSinghDev, 2021-12-15:
Ok, I faced a lot of weird reporting from Seachest_Format and SeaChest_Lite. 
On one of the drives I ran SeaChest_Lise --setSectorSize 4096 and after a restart the disk showed Logical Sector size: 4096 and Physical Sector size: 32768 which was weird. 

I then created a boot drive and --formatUnit 4096 --fastFormat 1. After the command the drive always showed 512 as the logical sector size but after reboot it showed 4096. 
I am attaching the releavent outputs 
[After.txt](https://github.com/Seagate/ToolBin/files/7718120/After.txt)
[AfterReboot.txt](https://github.com/Seagate/ToolBin/files/7718121/AfterReboot.txt)
[Before.txt](https://github.com/Seagate/ToolBin/files/7718122/Before.txt)
[EraseResults.txt](https://github.com/Seagate/ToolBin/files/7718123/EraseResults.txt)

vonericsen, 2021-12-15:
@JSinghDev,
Thanks for all the output.

This scenario is definitely odd. I don't see anything in the output that you provided that seem suspect to cause this.
Once the drive completes the format, it is supposed to report the new sector size, which has been my experience. 
A reboot is recommended afterwards anyways just to make sure all OS, drivers, etc can properly pick up the change....but SeaChest reads the information directly from the drive, which should report the new size immediately once it has finished (otherwise it reports format in progress until completed).

Can you confirm a couple things for me?

1. Is this controller configured in RAID mode?
2. Is this drive setup as a "passthrough" drive? (Assuming in RAID mode, not JBOD mode)
3. When you ran the fast format, how long did you wait before trying anything else? and about how long did it take SeaChest to output the "Successfully started format unit!"?

I have an older 8405 HBA and some other newer ones (don't remember models) that I can do a little testing with to see if I can repeat this, but it will take a bit before I can get that setup.

JSinghDev, 2021-12-16:
- All of my drives are brand new 4TB 512e/4kn drives. 
- The controller is running in HBA mode and not in RAID mode. The back plane is SGPIO and there is no SAS expander. 
- The command didn't take long, about 30 seconds in fast format.

I would like to add another thing that I saw with one of the drives when I was facing this issue. I ran the formatUnit command with fastFormat 0. Furthermore, when I realized the mistake, I cancelled the command and the drive went into a faulted state (03h I guess). After that, when I ran the SeaChest_Format - d .... -i command, it showed that the device did not show the formatUnit command in the supported features list, so I could not rerun the command. I tried rebooting, but it was the same.  This was in windows

I finally resolved that issue by going into the controller setup during boot and performing the low level format from there. Then post completion, the drive was behaving normally. 
I do not know if this is normal behavior, so I am just describing what I saw.

vonericsen, 2021-12-18:
Thanks for the extra detail @JSinghDev!

When you interrupted the format the first time, this puts the drive into "Format Corrupt" state. In this mode a lot of commands that SeaChest uses to detect drive features do not complete properly (even if the drive does support the command). This is because in format corrupt state certain commands are not available, but you should be able to send a new format to clear it and get it back to normal. This part makes sense.

When you did the fast format under Linux is where things seem odd to me since it seemed to take a reboot to show the new sector size. I don't know why this would be the case and nothing in what you've shared gives any indication of the drive or software having an issue, which is the odd part.

The next step will be for us to try and repeat the problem as best we can to debug it.

vonericsen, 2022-05-24:
I reread this thread and I think the reason format unit was not showing in the features is due to the drive being format corrupt (interrupting fast format mode 0).
The way the software looks for that to add to the list relies on a command I have recently observed reporting "format corrupt", so I added an additional message to the features list if this is seen.
I accidentally tagged #22 for this instead of #21 when I pushed it, hence this comment being added now.

The format unit command *should* always be supported by a SAS/SCSI drive, but that code also gets run for SCSI translators which are not known to support all translations, especially on USB, so I will need to figure out a better long term solution in the future, but I think this will help a little for now.
When this shows up in the output, it should inform that a format is necessary to use the drive.

The format done in the controller's BIOS was probably this same format unit command to get the drive back to a fully functioning state again.

closed 2023-03-06 by vonericsen