Seagate/ToolBin #22: FormatUnit fails

DavidXanatos, 2022-01-21:
I just got 5 new ST10000NM0578 E drivers
it seams all of them are set to 4096 sector size,
I want to use them on an older raid controller that supports only 512

So i attached them to a newer one a Adaptec ASR71605, set them to pass through 
and tried running 
```
SeaChest_Format_x64_windows_R.exe -d arc:0:0:0 --formatUnit 512 --fastFormat 1 --confirm this-will-erase-data
```
but unfortunately it returns en error

```
==========================================================================================
 SeaChest_Format - Seagate drive utilities - RAID Enabled - NVMe Enabled
 Copyright (c) 2014-2021 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Format Version: 2.3.1-2_2_3 X86_64
 Build Date: Jun 17 2021
 Today: Fri Jan 21 15:39:47 2022        User: Admin (admin)
==========================================================================================

arc:0:0:0 - ST10000NM0578 E - my serial number i guess - SCSI
Format Unit
Format Unit Failed!
```

Now this is not cool, I bought them explicitly to use them with that older controller,

Any ideas what may be failing here? How can I get them to switch to 512

when I run
```SeaChest_Format_x64_windows_R.exe -d arc:0:0:0 -i --showSupportedFormats```

it tels me
```
==========================================================================================
 SeaChest_Format - Seagate drive utilities - RAID Enabled - NVMe Enabled
 Copyright (c) 2014-2021 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Format Version: 2.3.1-2_2_3 X86_64
 Build Date: Jun 17 2021
 Today: Fri Jan 21 15:35:04 2022        User: Admin (admin)
==========================================================================================

arc:0:0:0 - ST10000NM0578 E - my serial number i guess - SCSI
        Vendor ID: IBM-ESXS
        Model Number: ST10000NM0578 E
        Serial Number: my serial number 
        Firmware Revision: ECH8
        World Wide Name: 5000C500C9C9A013
        Drive Capacity (TB/TiB): 9.93/9.03
        Temperature Data:
                Current Temperature (C): 43
                Highest Temperature (C): Not Reported
                Lowest Temperature (C): Not Reported
        Power On Time:  1 day 4 hours 54 minutes
        Power On Hours: 28.90
        MaxLBA: 2424569855
        Native MaxLBA: Not Reported
        Logical Sector Size (B): 4096
        Physical Sector Size (B): 4096
        Sector Alignment: 0
        Rotation Rate (RPM): 7200
        Form Factor: 3.5"
        Last DST information:
                Time since last DST (hours): 1.90
                DST Status/Result: 0x1
                DST Test run: 0x1
        Long Drive Self Test Time:  20 hours
        Interface speed:
                Port 0 (Current Port)
                        Max Speed (GB/s): 12.0
                        Negotiated Speed (Gb/s): 6.0
                Port 1
                        Max Speed (GB/s): 12.0
                        Negotiated Speed (Gb/s): Not Reported
        Annualized Workload Rate (TB/yr): 0.00
        Total Bytes Read (KB): 516.10
        Total Bytes Written (B): Not Reported
        Encryption Support: Self Encrypting
        Cache Size (MiB): Not Reported
        Read Look-Ahead: Not Supported
        Non-Volatile Cache: Enabled
        Write Cache: Disabled
        SMART Status: Good
        ATA Security Information: Not Supported
        Firmware Download Support: Full, Segmented, Deferred
        Number of Logical Units: 1
        Specifications Supported:
                SPC-5
                SAM-5
                SAS-3
                SPL-3
                SPC-4
                SBC-3
        Features Supported:
                Power Consumption
                Protection Type 1
                Protection Type 2
                TCG
                Application Client Logging
                Self Test
                Automatic Write Reassignment [Enabled]
                Automatic Read Reassignment [Enabled]
                EPC
                Informational Exceptions [Mode 6]
                Translate Address
                Seagate Remanufacture
                Seagate In Drive Diagnostics (IDD)
                Format Unit
                Fast Format
                Sanitize
                Storage Element Depopulation
        Adapter Information:
                Vendor ID: Not available.
                Product ID: Not available.
                Revision: Not available.

Supported Logical Block Sizes and Protection Types:
---------------------------------------------------
  * - current device format
PI Key:
  Y - protection type supported at specified block size
  N - protection type not supported at specified block size
  ? - unable to determine support for protection type at specified block size
Relative performance key:
  N/A - relative performance not available.
  Best
  Better
  Good
  Degraded
--------------------------------------------------------------------------------
 Logical Block Size  PI-0  PI-1  PI-2  PI-3  Relative Performance  Metadata Size
--------------------------------------------------------------------------------
                512     Y     ?     ?     N                   N/A            N/A
                520     Y     ?     ?     N                   N/A            N/A
                528     Y     ?     ?     N                   N/A            N/A
*              4096     Y     ?     ?     N                   N/A            N/A
               4112     Y     ?     ?     N                   N/A            N/A
               4160     Y     ?     ?     N                   N/A            N/A
--------------------------------------------------------------------------------
NOTE: Device is not capable of showing all sizes it supports. Only common
      sizes are listed. Please consult the product manual for all supported
      combinations.
NOTE: This device supports protection information (PI) (a.k.a. End to End protection).
        Type 0 - No protection beyond transport protocol
        Type 1 - Logical Block Guard and Logical Block Reference Tag
        Type 2 - Logical Block Guard and Logical Block Reference Tag (except first block)
                 32byte read/write CDBs allowed
      Not all forms of PI are supported on all sector sizes unless otherwise indicated
      in the device product manual.
NOTE: This device supports Fast Format. Fast format is not instantaneous and is used for
      switching between 5xx and 4xxx sector sizes. A fast format may take a few minutes or longer
      but may take longer depending on the size of the drive. Fast format support does not necessarily
      mean switching sector sizes AND changing PI at the same time is supported. In most cases, a
      switch of PI type will require a full device format.
      Fast format mode 1 is typically used to switch from 512 to 4096 block sizes with the current
          PI scheme.
```

So it should be working but its not it just fails, please help me to resolve this issue.

vonericsen, 2022-01-21:
Hi @DavidXanatos,

Can you provide the full verbose output?
The command will be something like this 
`SeaChest_Format_x64_windows_R.exe -d arc:0:0:0 --formatUnit 512 --fastFormat 1 --confirm this-will-erase-data -v4 > verbose.txt`
Then attach that output verbose file so I can look at it and figure out which command is failing in the process.

DavidXanatos, 2022-01-21:
sure her you go:

``` 
==========================================================================================
 SeaChest_Format - Seagate drive utilities - RAID Enabled - NVMe Enabled
 Copyright (c) 2014-2021 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Format Version: 2.3.1-2_2_3 X86_64
 Build Date: Jun 17 2021
 Today: Fri Jan 21 18:34:20 2022	User: Admin (admin)
==========================================================================================
Sending SCSI Inquiry

  CDB:

        0  1  2  3  4  5  
  0x00 12 00 00 00 60 00 

Sending command with send_IO

  Sense Data Buffer:

        0  1  2  3  4  5  6  7  
  0x00 00 00 00 00 00 00 00 00 

Sense Key: 0h = No Error
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
Information: 0000000000000000h
Command Specific Information: 0000000000000000h
Command Time (ms): 1.58

	  Data Buffer being returned:

        0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F  
  0x00 00 00 07 32 9F 01 10 02 49 42 4D 2D 45 53 58 53   ...2....IBM-ESXS
  0x10 53 54 31 30 30 30 30 4E 4D 30 35 37 38 20 45 20   ST10000NM0578 E 
  0x20 45 43 48 38 5A 50 57 30 37 35 41 34 45 43 48 38   ECH8ZPW075A4ECH8
  0x30 45 43 48 38 45 43 48 38 00 20 00 A2 0C 60 20 E0   ECH8ECH8. ...` .
  0x40 04 60 04 C0 00 00 00 00 00 00 00 00 00 00 00 00   .`..............
  0x50 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................

Inquiry returning: SUCCESS

Sending SCSI Report LUNs

  CDB:

        0  1  2  3  4  5  6  7  8  9  A  B  
  0x00 A0 00 00 00 00 00 00 00 00 08 00 00 

Sending command with send_IO

  Sense Data Buffer:

        0  1  2  3  4  5  6  7  
  0x00 00 00 00 00 00 00 00 00 

Sense Key: 0h = No Error
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
Information: 0000000000000000h
Command Specific Information: 0000000000000000h
Command Time (ms): 2.35

	  Data Buffer being returned:

        0  1  2  3  4  5  6  7  
  0x00 00 00 00 08 00 00 00 00                           ........

Report LUNs returning: SUCCESS
[... 250 lines cut ...]
	  Data Buffer being sent:

        0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F  
  0x00 00 28 00 10 01 00 00 10 FF FF FF FF FF FF FF FF   .(..............
  0x10 00 00 00 00 00 00 02 00                           ........

Sending command with send_IO

  Sense Data Buffer:

        0  1  2  3  4  5  6  7  
  0x00 00 00 00 00 00 00 00 00 

Sense Key: 0h = No Error
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
Information: 0000000000000000h
Command Specific Information: 0000000000000000h
Command Time (ms): 25.26

Mode Select 10 returning: OS PASSTHROUGH FAILURE

Format Unit Failed!

```

vonericsen, 2022-01-21:
Thanks for the verbose info.
The mode select 10 command is what sets the blocksize at the end, and the data all looks correct, but the RAID API being used returned an error, so I don't see if the drive responded with any specific error (Which is why it returned OS_PASSTHROUGH_FAILURE).
It is possible that the RAID blocked the command, but it doesn't give us that information.

Since you configured the drive as a passthrough device, does it show up in the `-s` output with a handle beginning with `PD`? Or does it only show with the `arc:0:0:0` handle? If it shows with `PD?`, can you also try it with that handle?

This also seems similar to #21 who also tried this on an Adaptec controller from Windows, and ended up having it work in Linux. The USB boot key should work and I believe it will find this device with a `/dev/sg?` handle if you try that.

DavidXanatos, 2022-01-21:
no it only shows arc, the be exact I did not configure anything, the controller default is to just pass the disk through unless its initialized.
And the device manager detects the drive properly i.e. it shows it as what it is a not some generic raid device like raid volumes are shown.

It is not showed with a PD? handle though :/

I could try one of the on board SAS ports, but IIRC those are quite old and quite probably don't support 4k sector drives, or is that irrelevant when only trying to send a few commands, I could configure those ports to AICH so no raid or anything.

Should I try that? before I try linux...

vonericsen, 2022-01-21:
I have not used an old SAS controller in a while, but to the best of my knowledge it should not care as long as the OS supports the 4k size. I would give it a try. 
The 4k sector size affects only read and write commands. Even XP had some 4k drive support as that is what USB drives emulated to get past the 2TB limit back then. So these other commands should not be affected.

I know some RAID controllers can be picky about what they allow or do not allow, but it could still be a weird thing from the drive...I do not know since the API did not get the raw drive response in the sense data which could help me figure it out better.

If you switch it and still do not see them show up, can you share what you do see in the `-s` output?

After that the next best thing I can recommend is trying Linux to see if it is able to communicate better. The SeaChest package on seagate.com includes a exe to write a Linux to a USB that can be booted and should come up and dump a list of what drives were found in the system. All SeaChest commands will be available from that command line too and the only major difference should be the device handle that is used.

DavidXanatos, 2022-01-22:
the onboard controller does not seam to see the drive at all :(
I'll try the linux option now , you mean this tool right: SeaChestUtilities\USB boot maker\..

DavidXanatos, 2022-01-22:
unfortunately it does not work with the linux tool eider
[image]

DavidXanatos, 2022-01-24:
@vonericsen does the screenshot tells you something helpful?

vonericsen, 2022-01-24:
Thanks for the output.
The drive response is saying it does not like the request to change to a 512B sector.
I don't know why though.

I'll see if I can figure out who might know why that is happening.
For a long time drives were sold as either 512 or 4k logical sector size and you couldn't switch between them, then fast format was developed to allow this. 
This drive does not seem too old, so it might be some other unique firmware configuration or something else may be going on.

DavidXanatos, 2022-01-24:
The drive is IBM labeled so perhaps they have some customization in the firmware :(

It would be great if the issue could be resolved.

But I think I'll just take that as a hint to stock up on one or two more modern RAID controller's that can handle 4k, that has much better performance then emulated anyways.

vonericsen, 2022-01-24:
Hi @DavidXanatos,

I asked one of the firmware team leads about this and this IBM firmware is locked to 4k sector sizes on this product. I do not know if other products with IBM firmware have this same limitation or not.
Standard OEM drives do not have these limitations since they are not dictated by specific customer requirements and contracts.
Sorry for the sad news!

DavidXanatos, 2022-01-24:
Well I saved almost 50% on these drives all new <100h work so I can afford a couple raid controllers for what I saved ;)
Thanks for the help.

closed 2022-01-24 by DavidXanatos

vonericsen, 2022-01-24:
That sounds like a good deal! 👍 

I'm going to add an additional warning to the "showSupportedFormats" output to mention that other customer firmwares may have additional restrictions on what sector sizes are allowed. Still working out the details on that messaging, but it will roll into the next update to these tools.