Seagate/ToolBin #25: SeaChest_Lite does not "see" any Exos ST16000NM001G connected to an Adaptec 71605 or 71605E.

LocoLati, 2022-04-26:
### Problem

SeaChest_Lite:  Latest version.

The software does not "see" any Exos ST16000NM001G connected to an Adaptec 71605 or 71605E.  Argument --scan seem not to work as usual.

I understand that this may be difficult if the drives are configured in a RAID set, but these drives a connected a single drives even in RAW PASS Thru mode.

BTW, is there a list of supported controllers available?

### Expected behavior

The software should work as nice as when Exos ST16000NM001G are connected to other controllers.

### How to reproduce

Brand new factory sealed Exos ST16000NM001G connected to 71605 or 71605E.  Not a single drive (out of more than 10) was seen by the software.  In order to change 512e to 4Kn I had to connect every drive to another controller, change sector size and reconnect to the Adaptec controller.  Then the drives can be accessed through Windows - but still not visible to SeaChest_Lite.

### Deployment information

Windows 10 Workstation, Windows 10 Enterprise, Adaptec 71605 or 71605E.  All latest versions and latest drivers.

vonericsen, 2022-05-02:
Hi @LocoLati,
Can you confirm the version of SeaChest_Lite that was used?
We recently pushed an updated version on Seagate.com but it is not yet pushed in this repository that may fix the scan issue. We had a very similar problem reported and it was due to CSMI (common storage management interface) being enabled in the driver and a bug causing a crash with this driver (Intel and AMD drivers we've tested do not seem affected). The bug is not completely resolved, but the scan is functional by skipping the CSMI scan on the driver that manages this HBA (arcsas).

Unfortunately we do not have a list of controllers supported and do not have the ability to create a list like that right now. I have a project to start working on better RAID support, but it will take quite a bit of time to get fully functional. When I work on this project I do plan to at least list the names of the supported drivers which will get closer to a list of what is supported.

Rungthida, 2022-05-03:
Hi @LocoLati,
"SeaChest Utilities" was already updated last week and "SeaChest Lite" was just updated today. Please try the current  version on Seagate.com and let us know if this version solves your issue. Thanks,

LocoLati, 2022-05-04:
Hi @vonericsen
Sorry for my delay.  Have been out for personal repair. ;-)
Here is the information you requested:

    SeaChest_Lite Version: 1.5.0-2_2_3 X86_64  -  Build Date: Jun 17 2021

Hope this helps?
Sorry me asking for a list you do not have at hand.  I asked only because it makes testing and verification more easier

Hi @Rungthida
At the moment all HDDs are in a production environment, so I feel unhappy testing with new software.  As soon as I get new drives I will report results.

Thanks to all for your help.

BTW Can you provide a hint why I get the message 
  "Setting sector size not supported on this device" 
on some of these drives?
As long as I do not fully understand this I am unsure to report a bug:

Let me supply some more data extracted from the screenshots:

SeaChest_Lite.EXE --device PD1 --showSupportedFormats
...
*               512     Y     N     N     N                   N/A            N/A
               4096     Y     N     N     N                   N/A            N/A

SeaChest_Lite.EXE --device PD1 --setSectorSize 4096 --confirm this-will-erase-data
...
Set Sector Size to 4096
...
Setting sector size not supported on this device

vonericsen, 2023-03-06:
@LocoLati,
I believe we have corrected the bug for both scan and the set sector size as of the most recent version in this repo (1.9.0). 
The previous 1.6.0 may have also corrected some of this, but may have had a CSMI bug. 1.9.0 includes a lot of fixes that may have caused a crash while scanning and caused the original issue you filed.

I'm closing this for now since we believe this has been fixed, but please feel free to reopen this issue if the problem still seems to persist with these newer versions.

closed 2023-03-06 by vonericsen