Seagate/ToolBin #27: Helps and docs don't explain how to pass a [range]

DevOpsCow, 2022-12-21:
### Problem

## Background:  

Win11 keeps doing something (sleep HDDs? even when set to never...)  and kills long running SeaChest operations: SMART long, erase, etc.

## The issue:  

I'm trying to erase a couple of HDDs (to sell away) but the erase command stops after ~20 minutes. So I went to use the `openSeaChest_Erase --writeSameRange`  option to perform the erase in steps.

The problem is neither the `--help` or the documents in this repository explain how the range has to be formatted.

I tried several options, including but not limited to:
```
openSeaChest_Erase -d PD1 --writeSameRange 1054894066-2109788130
openSeaChest_Erase -d PD1 --writeSameRange=1054894066-2109788130
openSeaChest_Erase -d PD1 --writeSameRange "1054894066 2109788130"
openSeaChest_Erase -d PD1 --writeSameRange 1054894066,2109788130
openSeaChest_Erase -d PD1 --writeSameRange 1054894066;2109788130
openSeaChest_Erase -d PD1 --writeSameRange [1054894066-2109788130]
```

## Request

It would be great if someone could:
1. Give me the example of how to format the LBA range to erase
2. Update the `--help` on the commands that use `[range] ` with an example for future users
3. Update the readme of the tools in the repo as well :)

Thanks!

### Expected behavior

The `--help`  argument for `openSeaChest_Erase` and other tools that use `[range]` should provide an example on how to format the range.

### How to reproduce

Run the follwoing command, (don't use the `--confirm` yet, the error will pop up right away)
```
H:\ToolBin-master\openSeaChest\bin-build\22.07.26\Win64>openSeaChest_Erase -d PD1 --writeSameRange 1054894066-2109788130

Error in option --writeSameRange. Invalid argument given '1054894066-2109788130'.
Use -h option to view command line help
```

Try any other combination of formats for the range: quotes, brackets, comma, colon, semicolon, etc

Finally, check the help, it doesn't provide a guideline or example on how to format the range:
```
        --writeSameRange [range]
                Specify a range to writesame to. Use this option with the
                writeSame option in order to begin a write same operation.
```

### Deployment information

Win 11 (boo!)
Local PC
SATA Drives attached to integrated Intel Chipset controller

vonericsen, 2022-12-21:
Hi @DevOpsCow,

Thank you for reporting this. I will look at updating the help to be a bit more descriptive for the range.
This is how it is supposed to be used:

`openSeaChest_Erase -d <handle> --writeSameStart x --writeSameRange y --confirm this-will....`

If you give a start, but no range, openSeaChest will automatically assume to go to the end of the drive.

> Win11 keeps doing something (sleep HDDs? even when set to never...) and kills long running SeaChest operations: SMART long, erase, etc.

This is an unfortunate thing that we've done everything we can to try and prevent from happening, but it is not always possible to stop Windows from doing this. You will likely keep encountering this unfortunately. FYI, Linux does this sometimes as well, but it is less common. Sometimes the OS decides "Hey I want to talk to that, but for some reason I cannot" and then it will interrupt these operations one way or another (usually with a reset). I have seen a few cases where this only happens when trying things on multiple drives in parallel, but single drive at a time works as expected.

You can use the `--overwrite` erase instead. This just has the software issue each write command instead of letting the firmware do it. Since this is managed by the software it should complete without error as the commands are much quicker to complete. It may be slightly slower than a firmware managed erase though. Generally on HDDs the difference is not huge unless they are attached to a slow bus like USB 2.0
The format of this is basically the same as the write same option:

`openSeaChest_Erase -d <handle> --overwrite x --overwriteRange y --confirm this-will...`

If you have a USB to SATA adapter, those do not often interrupt these kinds of things, so that might work for you as well if you wanted to try write same. On a USB to SATA adapter, you should be able to use Sanitize overwrite which cannot be interrupted by anything. It continues even after a power cycle.

`openSeaChest_Erase -d <handle> --sanitize overwrite --confirm this-will...`

Windows has blocked the sanitize commands since Windows 8 on SATA adapters, but Windows 11 does mention not blocking it on NVMe data drives....I have not tested to see if it is allowed on SATA now too. This block only exists in normal Windows. If you booted to the recovery environment (PE) these commands are not blocked and many other background services are turned off so other long operations may not be interrupted.

DevOpsCow, 2022-12-21:
Thank you for the extra info on the win11 possible issue. Great to know booting to the Win recovery might help if I need to do other kind of long operations (e.g. the long SMART)

And thanks for the alternate solution, I'm already doing the `--overwrite` and it seems to be working (already passed the original "stop" LBA)

vonericsen, 2023-03-06:
I'm closing this as I have updated this repo to the latest versions of the tools that include examples that I pushed earlier.
The examples in openSeaChest also carry over to closed SeaChest 😄 

Please feel free to reopen this issue if there is more we need to do on this issue.

closed 2023-03-06 by vonericsen