Seagate/ToolBin #28: SeaChest-lite SetSectorSize does nothing

johnteh00, 2023-05-30:
### Problem

When I try to change the sector size on a Exos 16 to 4kn nothing happens.
I start the process with the arguments"-d PD3 --setSectorSize 4096 --confirm this-will-erase-data"
The warnings and the countdown appear. After the Countdown has hit "0" the consol immediately jumps back out into the promt "C:\" 
I confirm, before and after restart, with --showSupportedFormats that  nothing has indeed changed and the drive is still at 512

### Expected behavior

I would expect a progress indication or at least a significant pause for Low Level Format to take place

### How to reproduce

-

### Deployment information

Windows 10 Pro 10.0.19044

vonericsen, 2023-07-18:
Hi @johnteh00,

Can you run this command with the verbose option and save it to a file so I can see what the drive is doing when this happens?
`SeaChest_Lite -d <handle> --setSectorSize 4096 --confirm this-will-erase-data -v 4 > setSectorSize.txt`

johnteh00, 2023-07-28:
Thanks @vonericsen. Weirdly enough, trying it in verbose mode with your prompt produced: "unknown option -- v"; but this time the format worked. It asked me to add "-and-may-render-the-drive-inoperable" to the prompt. And then continued to operate correctly

vonericsen, 2023-07-28:
@johnteh00, that is really odd...
One thing to be aware of is single character options only use a single `-` so maybe it was complaining about that? It's weird that that would suddenly change behavior, so I'm going to do my best to look into it and see if I can figure out how to repeat what you observed.

Can you confirm the version of SeaChest_Lite you are using?

johnteh00, 2023-07-30:
I did run it as single dash command. Tried a couple of times. IDK why the error would make it into double dash. Also when calling the command without options, Verbose wasn’t listed under the possible options

vonericsen, 2023-07-31:
@johnteh00,
I forgot that SeaChest_Lite does not allow verbose mode, so that explains that error.
I have not been able to repeat the behavior you saw where it quit without doing anything, so I'm not sure what is going on there still. 
I'm doing a code review to make sure I don't miss what could have caused this behavior in case it's something that only happens sometimes.

manotroll, 2024-08-07:
==========================================================================================
 SeaChest_Lite - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Lite Version: 1.9.2-6_2_0 X86_64
 Build Date: Dec  1 2023
 Today: Wed Aug  7 20:00:30 2024	User: admin
==========================================================================================
Invalid argument: SeaChest_Lite
Invalid argument: 4
One or more of the options provided will perform a low-level format that cannot
be interrupted once started. All background software should be stopped, any filesystems
that are currently mounted should first be unmounted in order to reduce the risk of
interruption. Do not attempt these operations on multiple devices at the same time
to ensure the best possible outcome. Many controllers/drivers/HBAs cannot handle these
operations running in parallel without issuing a device reset.
		There is a risk when performing a low-level format/fast format that may
		make the drive inoperable if it is reset at any time while it is formatting.
		WARNING: Any interruption to the device while it is formatting may render the
		         drive inoperable! Use this at your own risk!
		WARNING: Set sector size may affect all LUNs/namespaces for devices
		         with multiple logical units or namespaces.
		WARNING (SATA): Do not interrupt this operation once it has started or 
		         it may cause the drive to become unusable. Stop all possible background
		         activity that would attempt to communicate with the device while this
		         operation is in progress
		WARNING: It is not recommended to do this on USB as not
		         all USB adapters can handle a 4k sector size.
		WARNING: Disable any out-of-band management systems/services/daemons
		         before using this option. Interruptions can be caused by these
		         and may prevent completion of a sector size change.

If you wish to cancel this operation, press CTRL-C now to exit the software.

30
29
28
27
26
25
24
23
22
21
20
19
18
17
16
15
14
13
12
11
10
 9
 8
 7
 6
 5
 4
 3
 2
 1
 0

\\.\PhysicalDrive2 - ST4000DM005-2DP1 - WGY0AZKF - 0001 - SCSI

You must add the flag:
"this-will-erase-data-and-may-render-the-drive-inoperable" 
to the command line arguments to run a set sector size operation.

e.g.: SeaChest_Lite -d PD<#> --setSectorSize 4096 --confirm this-will-erase-data-and-may-render-the-drive-inoperable

		There is an additional risk when performing a low-level format/fast format that may
		make the drive inoperable if it is reset at any time while it is formatting.
		WARNING: Any interruption to the device while it is formatting may render the
		         drive inoperable! Use this at your own risk!
		WARNING: Set sector size may affect all LUNs/namespaces for devices
		         with multiple logical units or namespaces.
		WARNING (SATA): Do not interrupt this operation once it has started or 
		         it may cause the drive to become unusable. Stop all possible background
		         activity that would attempt to communicate with the device while this
		         operation is in progress
		WARNING: It is not recommended to do this on USB as not
		         all USB adapters can handle a 4k sector size.
		WARNING: Disable any out-of-band management systems/services/daemons
		         before using this option. Interruptions can be caused by these
		         and may prevent completion of a sector size change.

vonericsen, 2025-08-01:
@manotroll,

We just updated the version of SeaChest_Lite in this repository. Can you try the new version to see if it behaves any differently?

Also, one thing I didn't catch before was that this drive is showing up as a SCSI device type but other information about this drive shows its SATA.
How is it attached to your system when you are running this? That may be why it wasn't working. We did do some updates in our low-level around some USB adapters which may help if that is how it is connected.

manotroll, 2025-08-02:
and a SAS controller the drive interface and SATA

vonericsen, 2025-08-12:
Which SAS controller are you using?