Seagate/ToolBin #30: Failed to set sector size!

clertes, 2023-09-18:
### Problem

I'm receiving error message `Failed to set sector size!` while trying to change the sector size of my ST2000NM017B devices from 512 (default) to 4096.

```
~/SeaChest/SeaChestLite/Lin64/Non-RAID/x86_64 # ./SeaChest_Lite_x86_64-alpine-linux-musl_static --scan
==========================================================================================
 SeaChest_Lite - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Lite Version: 1.9.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Mon Sep 18 17:13:07 2023	User: root
==========================================================================================
Vendor   Handle       Model Number            Serial Number          FwRev
ATA      /dev/sg0     ST2000NM017B-2TD103     XXXXXXXX               SN01
ATA      /dev/sg1     ST2000NM017B-2TD103     XXXXXXXX               SN01
ATA      /dev/sg2     ST2000NM017B-2TD103     XXXXXXXX               SN01
ATA      /dev/sg3     ST2000NM017B-2TD103     XXXXXXXX               SN01
ATA      /dev/sg4     ST2000NM017B-2TD103     XXXXXXXX               SN01
ATA      /dev/sg5     ST2000NM017B-2TD103     XXXXXXXX               SN01
ATA      /dev/sg6     ST2000NM017B-2TD103     XXXXXXXX               SN01
ATA      /dev/sg7     ST2000NM017B-2TD103     XXXXXXXX               SN01
ATA      /dev/sg8     ST2000NM017B-2TD103     XXXXXXXX               SN01
ATA      /dev/sg9     ST2000NM017B-2TD103     XXXXXXXX               SN01`
```

512 and 4096 should be supported:

```
~/SeaChest/SeaChestLite/Lin64/Non-RAID/x86_64 # ./SeaChest_Lite_x86_64-alpine-linux-musl_static --device /dev/sg0 --showSupportedFormats
==========================================================================================
 SeaChest_Lite - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Lite Version: 1.9.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Mon Sep 18 17:14:57 2023	User: root
==========================================================================================

/dev/sg0 - ST2000NM017B-2TD103 - XXXXXXXX - SN01 - ATA

Supported Logical Block Sizes and Protection Types:
---------------------------------------------------
  * - current device format
PI Key:
  Y - protection type supported at specified block size
  N - protection type not supported at specified block size
  ? - unable to determine support for protection type at specified block size
Relative performance key:
  N/A - relative performance not available.
  Best
  Better
  Good
  Degraded
--------------------------------------------------------------------------------
 Logical Block Size  PI-0  PI-1  PI-2  PI-3  Relative Performance  Metadata Size
--------------------------------------------------------------------------------
*               512     Y     N     N     N                   N/A            N/A
               4096     Y     N     N     N                   N/A            N/A
--------------------------------------------------------------------------------
```

Any advices?

### Expected behavior

Set sector size to 4096

### How to reproduce

Error message:

```
~/SeaChest/SeaChestLite/Lin64/Non-RAID/x86_64 # ./SeaChest_Lite_x86_64-alpine-linux-musl_static --device /dev/sg0 --setSectorSize 4096 --confirm this-will-erase-data-and-may-render-the-drive-inoperable
==========================================================================================
 SeaChest_Lite - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Lite Version: 1.9.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Mon Sep 18 17:08:36 2023	User: root
==========================================================================================
One or more of the options provided will perform a low-level format that cannot
be interrupted once started. All background software should be stopped, any filesystems
that are currently mounted should first be unmounted in order to reduce the risk of
interruption. Do not attempt these operations on multiple devices at the same time
to ensure the best possible outcome. Many controllers/drivers/HBAs cannot handle these
operations running in parallel without issuing a device reset.
		There is a risk when performing a low-level format/fast format that may
		make the drive inoperable if it is reset at any time while it is formatting.
		WARNING: Any interruption to the device while it is formatting may render the
		         drive inoperable! Use this at your own risk!
		WARNING: Set sector size may affect all LUNs/namespaces for devices
		         with multiple logical units or namespaces.
		WARNING (SATA): Do not interrupt this operation once it has started or
		         it may cause the drive to become unusable. Stop all possible background
		         activity that would attempt to communicate with the device while this
		         operation is in progress
		WARNING: It is not recommended to do this on USB as not
		         all USB adapters can handle a 4k sector size.
		WARNING: Disable any out-of-band management systems/services/daemons
		         before using this option. Interruptions can be caused by these
		         and may prevent completion of a sector size change.

If you wish to cancel this operation, press CTRL-C now to exit the software.
 0

/dev/sg0 - ST2000NM017B-2TD103 - XXXXXXXX - SN01 - ATA
Set Sector Size to 4096

WARNING: It is not recommended to do this on USB enclosures as not
         all USB adapters can handle a 4k sector size.
WARNING (SATA): Do not interrupt this operation once it has started or
         it may cause the drive to become unusable. Stop all possible background
         activity that would attempt to communicate with the device while this
         operation is in progress
Press CTRL-C to cancel this operation before the timer runs out.
  0
Setting the drive sector size quickly.
Please wait a few minutes for this command to complete.
It should complete in under 5 minutes, but interrupting it may make
the drive unusable or require performing this command again!!
WARNING: Unable to erase MBR. If unable to write a partition after this operation, erase the first sector of the device
         and the last sector (max LBA) then try creating new partitions again.
Failed to set sector size!
```

### Deployment information

Ubuntu 18.04 LTS

vonericsen, 2024-02-20:
@clertes,
I'm sorry I am just now seeing this issue.

When this error happens, it means either the drive reported a problem while trying to change the sector size, or (more likely) something within the operating system interrupted the sector size change with a reset.
This is why the tools come with so many different warning messages around this feature as it is not possible through any method I have found to stop the OS from issuing this reset.

The only advice I can give is to try it again, making sure all software running in the background is stopped and possibly trying this from a USB bootable environment.

vonericsen, 2025-08-01:
We have posted a new version of our tools which include some additional changes to minimize the possibility of interruptions during the sector size change.
The new changes open the drive handle with exclusive access before starting the format which seems to reduce the chance of this problem occurring.

I'm going to close this for now since we did make more improvements to software to minimize these resets from outside sources, but please feel free to reopen this or create a new issue if you keep running into this.

closed 2025-08-01 by vonericsen