Seagate/ToolBin #32: Seagate FireCuda 530 - Error: IDD is only supported on Seagate Drives.

Districtor, 2023-09-27:
### Problem

Input:

SeaChest_SMART_x64_windows -d PD4 --idd short --poll

Message:

 SeaChest_SMART - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_SMART Version: 2.3.1-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Wed Sep 27 19:17:20 2023        User: distr (admin)

\\.\PhysicalDrive4 - Seagate FireCuda 530 ZP1000GM30013 - 7VQ06XXX - SU6SM005 - NVMe
IDD is only supported on Seagate Drives.

### Expected behavior

Better error description.

### How to reproduce

See above.

### Deployment information

Windows 22H2 22621.2361

### Additional information

The same input works on:

ST14000NM001G
ST8000NM0055

vonericsen, 2024-02-20:
Hi @Districtor,

Sorry I'm just now seeing this issue.
IDD is an HDD only feature.

I agree the error message is not correct though and I will look into changing it to something else.

Districtor, 2024-02-20:
> Hi @Districtor,
> 
> Sorry I'm just now seeing this issue. IDD is an HDD only feature.
> 
> I agree the error message is not correct though and I will look into changing it to something else.

Thank you.

closed 2024-05-16 by Districtor