Seagate/ToolBin #34: format error - SeaChest_Release_3.5.2_03-29-2023__15-05-29.usbBootMaker

Theletha, 2024-03-13:
### Problem

can't format the drive via the program to continue, outputting error 164
[image]

### Expected behavior

format

### How to reproduce

try to press create

### Deployment information

win 11 x64 

### Additional information

tried 4gb, 8gb and 16gb stick but all give same issue

vonericsen, 2024-03-14:
Hi @Theletha,

Can you give me some details about the USB sticks you've tried?

Do you have a brand or model?

Were these USB drives previously formatted with a different file system (for example exfat or NTFS)?

Are you able to format them through Windows? You can do this by going into "This PC" then right clicking on the drive and selecting "Format". That will pop up a window where you can select file system, name, and other information to format with.

Theletha, 2024-03-16:
Tried Kingston sticks. I tried to pre format all before the installer tried exfat, ntfs, fat32 but no go.

vonericsen, 2024-03-19:
@Theletha,

We have been debugging and we think we have found the cause of the error and have been able to recreate the issue.
I will update this issue when the package has been updated on the Seagate.com site

Theletha, 2024-03-21:
Great news!

closed 2024-03-21 by Theletha