Seagate/ToolBin #36: Cannot Seem to run Seachest Utilities in Ubuntu

Jeremyltw, 2024-11-14:
### Problem

Hi, I have a server version of ubuntu and I'm trying to run the seachest commands in there. However, I can't seem to get it to run. I keep getting an error message SeaChest_PowerControl: command not found.

I've copied the files over (Linux/Non-Raid/x86_64) from my windows computer into my ubuntu. 

I've tried settting the path to a custom folder and copied some of the files into /usr/local/bin but nothing seems to work. Not sure if someone can help me on this please.

### Expected behavior

The commands should have been executed as per the guide provided.

### How to reproduce

Trying to execute any of the commands as per the guide results in the issue even with the files copied into /usr/local/bin

### Deployment information

Ubuntu 22.04.4 LTR

vonericsen, 2024-11-26:
Hi @Jeremyltw,

Sorry I did not see this issue earlier!

> I've copied the files over (Linux/Non-Raid/x86_64) from my windows computer into my ubuntu.

One thing that can happen when moving between Windows and Linux is often times the execute permission will be removed from the file.
You can check if the execute permission is set in the columns reported by `ls -la /usr/local/bin`
If the `x` is missing then you can add it with `chmod 0755 /usr/local/bin/SeaChest_*` which will add it to all the SeaChest files.
You can adjust the permissions if you want something other than 755, but that gives `rwx` to the user and `rx` to the group and others (like root).

If that does not work, can you share the output of `ls -la /usr/local/bin` so I can see if anything else looks off?

Jeremyltw, 2024-11-28:
Ok, that seemed to work, thanks!

vonericsen, 2024-12-02:
@Jeremyltw, 

Do you think that this is missing from documentation somewhere? I just want to make sure we have this resolved for future users as well and if there is a place you think it would be helpful to document this, I will look at getting it updated.

Jeremyltw, 2024-12-03:
@vonericsen yes, i think it would be great. to include into the existing documents for each SeaChest_ functions. Best to have it up front perhaps at where the Usage - Linux (run with sudo) is as a note to ensure that the permissions using chmod is done right