Seagate/ToolBin #37: Seagate Firecuda 530 on Thunderbolt 4 Acasis TBU405 - no sector size to 4096 alignment possible

Districtor, 2025-01-22:
### Problem

Asus Hero XIII on-board Thunderbolt 4 (Intel JHL8540) to Acasis TBU405 (Intel JHL7440) external enclosure. The TBU is recognized via TB Control Center, the Firecuda 530 is recognized as a Firecuda 530 and it can be interacted with normally via Seachest command-line utilities. Flashing the firmware in Win 11 Pro worked fine, since sector alignment does not work in Windows environment, Seaboot was used.

Seaboot (tinycorelinux) recognizes the drive as Firecuda 530 and it can be interacted with normally. 4096 sector size is shown as available.
However alignment fails with the error message shown on the screenshot.

[image]

### Expected behavior

Sector size set to 4096.

### How to reproduce

See screenshot.

### Deployment information

Seaboot (tinycorelinux).

Districtor, 2025-01-23:
Thank you for your fast reply.

Did not work, exact same error. Note: SeaChest_Lite, as per your example, does not support the  --nvmFormat option, SeaChest_NVMe was used in all instances.

Attached is the verbose 4 txt file for 4096, i do not think that SeaChest_Lite supports verbose.

[sector.txt](https://github.com/user-attachments/files/18528152/sector.txt)

vonericsen, 2025-01-24:
@Districtor,

Thanks for sharing the verbose output.

>Note: SeaChest_Lite, as per your example, does not support the --nvmFormat option, SeaChest_NVMe was used in all instances.

You are right! That was my mistake, I saw SeaChest_Lite at the top of the image and used that....oops.

I think I see the problem when I am walking through the code. 
There is validation that a LBA format identifier is within range of what the drive supports. It seems that a bug was introduced when we added some support for NVMe 2.0 specification which upped the maximum number of formats from 16 to 64.
I will make a change and test this. I can share a test build once I have verified it fixes the issue.

If you are comfortable using nvme-cli as another solution, this should be the equivalent of what SeaChest is running: `nvme format /dev/nvme2n1 --lbaf 1`
I do not remember if nvme-cli polls for progress so you may need to wait a few minutes for it to complete before removing power/rebooting.

I don't remember if nvme-cli is included in the USB image we use, but it might be. You could alternatively flash something like Ubuntu to a USB, install nvme-cli with apt, and then run this command.
On the USB (/dev/sde1 in your image) there are some folders you can check to see if it's there. You can use `ls ./<folder>` to see what's in each and if it is present or not. Sometimes we include other tools when we build the image.

Districtor, 2025-01-24:
> I can share a test build once I have verified it fixes the issue.

Can i please test it? 

Thank you for your advice on nvme-cli, i checked and it is not included on the stick.

vonericsen, 2025-01-27:
@Districtor,
Glad you were able to get that working in nvme-cli!

> One thing I noticed with Seaboot: The main screen shows: 13-Apr-2020, yet the included SeaChest utilities show their build date
as: Sep 26 2024. Somewhat confusing.

I think that may be from a script in the image. I'll look into updating this to reduce the confusion on our next build.

vonericsen, 2025-01-27:
@shriya-deshmukh,
I believe I have fixed the format NVM issue in SeaChest/openSeaChest already.

Can you look into where the 13-Apr-2020 message is coming from on the USB?
We will want to update this at next build.

We will work together to make sure all the other included tools (like NVMe-cli) are also the latest versions before we build a new version.

Districtor, 2025-04-05:
If I may ask:

> I believe I have fixed the format NVM issue in SeaChest/openSeaChest already.

Has the issue been fixed, should I close this topic?

> We will work together to make sure all the other included tools (like NVMe-cli) are also the latest versions before we build a new version.

Has Seaboot been updated?

vonericsen, 2025-04-07:
@shriya-deshmukh,
Did this get published to the web yet?
I was waiting to close this for the web publication in case any other users came across this issue.

shriya-deshmukh, 2025-04-08:
@vonericsen 
The fix for the incorrect date and the update of the NVMe-cli tool in the USB boot has been checked in, but the new release of SeaChest USB Boot/CLI utility has not been published on the web yet.

vonericsen, 2025-04-08:
Thanks @shriya-deshmukh!

We are still verifying some changes in SeaChest/openSeaChest to make sure everything works as expected before we roll the new version out.
I think most of the issues are solved, but not sure on the exact timeline of the release just yet.

vonericsen, 2025-08-01:
The new builds I posted today resolve this in SeaChest and openSeaChest.
The USB bootable has been updated and will be on the website within a couple weeks.

Sorry this issue took so long to resolve...ran into numerous other issues getting this release completed.

Districtor, 2025-08-01:
Nice, those are great tools to have.