Seagate/ToolBin #38: seachest_info returns nothing as far as I can tell.

Stokestack, 2025-02-01:
### Problem

I have a new Seagate ST8000DM, hooked up to a computer running Linux Mint.

Running seachest_info on it returns no useful stats, regardless of options.

Examples:

```
sudo seachest_info -d /dev/sdc --deviceStatistics
==========================================================================================
 SeaChest_Info - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Info Version: 2.5.0-6_2_0 X86_64
 Build Date: Dec  1 2023
 Today: Sat Feb  1 02:23:41 2025	User: root
==========================================================================================

/dev/sg2 - 004-2U9188 -  {'s                 - 0001 - SCSI
===Device Statistics===
	* = condition monitored with threshold (RLEC Feature)
	! = monitored condition met (Requires Threshold to be set and comparison enabled)
	- = supports notification (requires log page thresholds to be supported)
 Statistic Name:                                              Threshold:       Value:  
```

```
sudo seachest_info -d /dev/sdc -i
==========================================================================================
 SeaChest_Info - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Info Version: 2.5.0-6_2_0 X86_64
 Build Date: Dec  1 2023
 Today: Sat Feb  1 02:15:37 2025	User: root
==========================================================================================

/dev/sg2 - 004-2U9188 -  {'s                 - 0001 - SCSI
	Vendor ID: ST8000DM
	Model Number: 004-2U9188      
	Serial Number: {'s
	Firmware Revision: 0001
	World Wide Name: Not Supported
	Drive Capacity (TB/TiB): 2.20/2.00
	Temperature Data:
		Current Temperature (C): Not Reported
		Highest Temperature (C): Not Reported
		Lowest Temperature (C): Not Reported
	Power On Time: Not Reported
	Power On Hours: Not Reported
	MaxLBA: 4294967295
	Native MaxLBA: Not Reported
	Logical Sector Size (B): 512
	Physical Sector Size (B): 512
	Sector Alignment: 0
	Rotation Rate (RPM): Not Reported
	Form Factor: Not Reported
	Last DST information:
		Not supported
	Long Drive Self Test Time: Not Supported
	Interface speed:
		Max Speed (MB/s): 5.00 (FAST-5)
		Negotiated Speed (MB/s): Not Reported
	Annualized Workload Rate (TB/yr): Not Reported
	Total Bytes Read (B): Not Reported
	Total Bytes Written (B): Not Reported
	Encryption Support: Not Supported
	Cache Size (MiB): Not Reported
	Read Look-Ahead: Not Supported
	Write Cache: Not Supported
	SMART Status: Unknown or Not Supported
	ATA Security Information: Not Supported
	Firmware Download Support: Not Supported
	Specifications Supported:
		SCSI
	Features Supported:
		None reported or an error occurred while trying to determine
		the features.
	Adapter Information:
		Adapter Type: USB
		Vendor ID: 067Bh
		Product ID: 2773h
		Revision: 0000h
```

### Expected behavior

Stats are populated.

### How to reproduce

Run the above command against a similar drive.

### Deployment information

Installed from binary builds in the repo. OS is Linux Mint 21.3

vonericsen, 2025-05-15:
Hi @Stokestack,

I'm sorry I didn't see this issue earlier!

My guess is that this adapter uses a different passthrough mechanism than SAT.
I'm guessing this for two reasons: 
1.  the adapter is only reporting 2TB of capacity  
2. I see the vendor ID matches `Prolific` which I do have a very old passthrough mechanism in our code we can try for this adapter.
3. our latest code does a lot of automated retries for SAT and none of them appear to have worked.

I have an automated test under the [openSeaChest project](https://github.com/Seagate/openSeaChest/).

If you are willing to run the tool `openSeaChest_PassthroughTest` that can give enough output for me to know if it is that passthrough or not.

You would need to run `openSeaChest_PassthroughTest -d <handle> --runPTTest --ptDriveHint ata --ptTypeHint legacyATA --enableLegacyATAPTTest | tee results.txt` then share the created results.txt file on here for me to review (You can attach it or copy the output to share).

If you have used this adapter with smartmontools successfully, can you share what options you used to get it working? That may also help.

Stokestack, 2025-05-16:
Thanks for the follow-up! At this point I just formatted the thing and put it into service. Next time I'll give this a try.

vonericsen, 2025-05-16:
@Stokestack,

Sounds good. I'm going to leave this open in case someone else has the same kind of problem with the same adapter.
Next time you bust it out to test a drive (even if it isn't a Seagate drive) feel free to run the test I mentioned and share the output.

We do our best to support everything we can, but sometimes USB adapters get funky and are difficult to troubleshoot (and often difficult for us to get the exact same model to test ourselves).

Stokestack, 2025-05-16:
Well, I for one appreciate your efforts!

vonericsen, 2025-08-01:
I'm closing this because I did look up the adapter IDs and saw how they are handled in smartmontools and routed this adapter to the prolific passthrough we have. This is available in the release I pushed to this repo today and the v25.05.2 build of openSeaChest.

If you run into issues, feel free to reopen this or create a new issue for us to look at!