Seagate/ToolBin #39: Seagate Firecuda 530 - No firmware update SU6SM100 possible

Districtor, 2025-06-29:
### Problem

Firmware update per Seatools (Windows) and SeaChest (CLI) reports successful operation, however, after a power cycle, even with the PSU turned off, the firmware remains at version SU6SM005.

Many users experience the same problem:

https://www.reddit.com/r/Seagate/comments/1lk9229/seagate_firecuda_530_firmware_su6sm100_release/

Tried on 3 different 530s, all with the same results. Tried --downloadMode auto and different versions of SeaChest_Firmware, problem remains.

C:\Temp\fw>SeaChest_Firmware_x64_windows -d PD5 --downloadFW FireCuda530_SU6SM100.bin --downloadMode deferred --activateFW
==========================================================================================
 SeaChest_Firmware - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Firmware Version: 4.0.0-6_2_0 X86_64
 Build Date: Dec  1 2023
 Today: Sun Jun 29 17:14:29 2025        User: admin
==========================================================================================

\\.\PhysicalDrive5 - Seagate FireCuda 530 ZP2000GM30013 - 7VR052XK - SU6SM005 - NVMe
.....
Firmware Download successful
Firmware Download time (ms): 884.02
Average time/segment  (ms): 12.76
Activate Time          (ms): 12.43
Firmware download complete. Reboot or run the --activateFW command to finish installing the firmware.
Firmware activation successful
The Operating system has reported that a power cycle is required to complete the firmware update

### Expected behavior

Update to SU6SM100 is successful and reported

### How to reproduce

See above.

### Deployment information

Windows 11 24H2 26100.4349

Districtor, 2025-06-30:
Asus ROG MAXIMUS XIII HERO, all 3 530s are connected via PCIe. All 3 drives are compatible (Part number) with SU6SM100 as per the included READMEFIRST-FireCuda530-SU6SM100-NVMe SSD firmware update.pdf.

[fwdlInfo.txt](https://github.com/user-attachments/files/20988026/fwdlInfo.txt)
[veryveryverboseUpdate.txt](https://github.com/user-attachments/files/20988025/veryveryverboseUpdate.txt)

> SeaChest_Firmware -d PDX --showFWDLInfo

Do you mean --fwdlInfo?

C:\Temp\fw>SeaChest_Firmware_x64_windows -d PD5 --fwdlInfo
==========================================================================================
 SeaChest_Firmware - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023

 Today: Tue Jul  1 00:20:52 2025        User: distr (admin)
==========================================================================================

\\.\PhysicalDrive5 - Seagate FireCuda 530 ZP2000GM30013 - 7VR052XK - SU6SM005 - NVMe
===Download Support information===
Model Number: Seagate FireCuda 530 ZP2000GM30013
Firmware Revision: SU6SM005
Modes Supported:
        Deferred (Requires activation command to activate)
Maximum Segment Size (512B Blocks): 4096
Minimum Segment Size (512B Blocks): 8
Recommended Segment Size (512B Blocks): 0
Buffer Offset required (Bytes): 4096
Firmware Slot Info:
        Slot 1 (Active): SU6SM005
        Slot 2: Not Available
        Slot 3: Not Available
        Slot 4: Not Available

C:\Temp\fw>SeaChest_Firmware_x64_windows -d PD5 --llInfo
==========================================================================================
 SeaChest_Firmware - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Tue Jul  1 00:22:40 2025        User: distr (admin)
==========================================================================================

\\.\PhysicalDrive5 - Seagate FireCuda 530 ZP2000GM30013 - 7VR052XK - SU6SM005 - NVMe

---Low Level tDevice information---
        ---Drive Info---
                media type: NVM
                drive type: NVMe
                interface type: NVMe
                zoned type: not zoned
                ---adapter info---
                        PCI/PCIe:
                        VendorID: 1BB1h
                        ProductID: 5018h
                        Revision: 0001h
                ---driver info---
                        driver name: stornvme
                        driver version string: 10.0.26100.4202
                                major ver: 10
                                minor ver: 0
                                revision: 26100
                                build number: 4202
                Namespace ID set as 1
                SCSI Version: 6
                ---Passthrough Hacks---
                        Passthrough type: SAT/system/none
                                ---SCSI Hacks---
                                ---NVMe Hacks---
                                        LIMPT (Limited passthrough capabilities. Only specific commands are allowed)
                                        Allowed commands:
                                                Identify controller
                                                Identify namespace
                                                Get Log Page
                                                Get Features
                                                Firmware Download
                                                Firmware Commit
                                                Vendor Unique (Commands effects log)
                                                Device Self Test
                                                Sanitize
                                                Set Features
                                                Security Send
                                                Security Receive
                                                Format - SES = user erase
                                                Format - SES = crypto erase
        ---OS Info---
                handle name: \\.\PhysicalDrive5
                friendly name: PD5
                minimum memory alignment: 4
                ---Windows Unique Info---
                        Primary handle opened
                        SCSI Address:
                                Port Number: 2
                                Path ID: 0
                                Target ID: 0
                                LUN: 0
                        os drive number: 5
                        SRB type: 1
                        Alignment Mask: 3h
                        IOCTL Type: Not Set
                        IOCTL Method: Force Direct
                        FWDL IOCTL Support:
                                Payload alignment: 4096B
                                Maximum Transfer size: 2097152B
                        Adapter reported max transfer size: 2097152B
                        Miniport FWDL IOCTL is supported
                        Volume bitfield: 40
                        Adapter Descriptor bustype: NVMe
                        Device Descriptor bustype: NVMe
                OS read-write recommended: true
                last recorded error: 0
                File system Info:
                        Active file system detected on device

Districtor, 2025-07-01:
Thank you for your ongoing efforts.

Just tried --firmwareSlot 1, "Firmware Download successful" reported, computer was shut down, firmware still at SU6SM005.

C:\Temp\fw>seaChest_Firmware_x64_windows -d PD5 --downloadFW FireCuda530_SU6SM100.bin --downloadMode deferred --firmwareSlot 1 --activateFW
==========================================================================================
 SeaChest_Firmware - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Firmware Version: 3.4.0-4_1_1 X86_64
 Build Date: Mar 27 2023
 Today: Tue Jul  1 20:32:41 2025        User: distr (admin)
==========================================================================================

\\.\PhysicalDrive5 - Seagate FireCuda 530 ZP2000GM30013 - 7VR052XK - SU6SM005 - NVMe
.....
Firmware Download successful
Firmware Download time (ms): 857.26
Average time/segment  (ms): 12.49
Activate Time          (ms): 12.43
Firmware download complete. Reboot or run the --activateFW command to finish installing the firmware.
Firmware activation successful
The Operating system has reported that a power cycle is required to complete the firmware update

...

NVMe     PD5          Seagate FireCuda 530 ZP2000GM30013 7VR052XK               SU6SM005

vonericsen, 2025-07-02:
I have figured out the root cause of the issue.

Back in Win10, there were limited flags to activate FW on NVMe. 
Win11 has had some API updates to support more methods, so what worked on Win10 is being interpreted differently now.
When I changed our code to use some of the new flags to activate the firmware, it took the update.
I did this as a quick hack while testing, so I need to come up with a better fix, but I will be working on that ASAP and will share a new/test build when I have one ready.

vonericsen, 2025-07-07:
Windows 10 has been tested with these changes, and it is still working as expected (didn't want to break anything already working).
Windows 11 23H2 is tested now and working.
I need to test Win11 24H2 and then we can roll this fix out.

Districtor, 2025-07-08:
Can confirm the beta works, drive is working without issues. Good work!

C:\Temp\fw>openSeaChest_Firmware -d PD7 --downloadFW FireCuda530_SU6SM100.bin --downloadMode deferred --activateFW
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2025 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 4.3.2 X86_64
 Build Date: Jul  8 2025
 Today: 20250709T000012 User: admin
==========================================================================================

\\.\PhysicalDrive7 - Seagate FireCuda 530 ZP1000GM30013 - 7VQ0353N - SU6SM005 - NVMe
.....
Firmware Download successful
Firmware Download time (ms): 630.08
Average time/segment  (ms): 9.57
Activate Time          (ms): 9.51
Firmware download complete. Reboot or run the --activateFW command to finish installing the firmware.
Firmware activation successful
New firmware version is SU6SM100

AtomicInternet2, 2025-07-10:
Can confirm working on Windows 11 24H2 26100.4652 with Seagate ZP1000GM30053 as well.  Thanks for the update!

ceedveed, 2025-07-10:
Thanks I tested v25.05.2-beta1 on Windows 11 24H2. For anyone else that is a newbie like me:

Downloaded [openSeaChest-v25.05.2-beta1-win-x64.zip](https://github.com/Seagate/openSeaChest/releases/download/v25.05.2-beta1/openSeaChest-v25.05.2-beta1-win-x64.zip)
Downloaded the firmware from: https://apps1.seagate.com/downloads/request.html
Put the firmware in the same folder as the beta.
Ran command `openSeaChest_Firmware.exe --scan` to understand which PDX was my drive. Mine was PD1.
Ran command `openSeaChest_Firmware.exe -d PD1 --downloadFW FireCuda530_SU6SM100.bin`
Rebooted PC.
Verified firmware with command `openSeaChest_Firmware.exe -d PD1 --fwdlInfo`
Also verified firmware with CrystalDiskInfo.

vonericsen, 2025-07-10:
If it helps anyone else, I did put together a [wiki page for those who are new to using CLI tools](https://github.com/Seagate/openSeaChest/wiki/Introduction-To-Command-Line-Tools).

Feel free to leave any feedback on what can be improved as I am happy to address it!

Districtor, 2025-07-10:
A little off-topic, but nvme-cli is another option to flash SU6SM100. Tried on one of my 530s and the update completed without errors.

nvme-cli can also set the sector size, see https://github.com/Seagate/ToolBin/issues/37

jdabreu, 2025-07-26:
Can we go from version SU6SM003 straight to this one?

closed 2025-07-30 by vonericsen

vonericsen, 2025-08-01:
@R34lt1m3,

Sorry I missed your earlier message! 

> Do you happen to have any official information from Seagate regarding what SU6SM100 specifically addresses?

Unfortunately, I don't have any documentation about this.

vonericsen, 2025-08-01:
@jdabreu

> Can we go from version SU6SM003 straight to this one?

I believe this is supported, but I haven't tried it myself. The package on Seagate.com included the in between SU6SM005 build which you could use in between if you want.
Usually when updates are posted like this it is possible to move to the latest firmware without needing to do to an in-between version.

This may not always be the case, but if the drive does not support this kind of version jump, then it will reject the download with an error. In some cases we can get enough info in our tools to say it's an error with needing some other in-between version, but not always...and usually not in Windows due to how Microsoft wrote their API.